riscv/riscv-cheri #623: allocate misa.y

tariqkurd-repo, 2025-06-06:
It makes sense to allocate misa.y, and set it when in cap ptr mode.

For a hybrid system, after reset, misa.i/e should be set initially, and to be mutually exclusive with misa.y.
Cap ptr mode entry will set misa.y and clear misa.i/e

Would we want misa.y to be directly writeable?
Could we allow that as a way to enter cap ptr mode?
COuld this be a discovery method for a CHERI CPU?

I'm not sure about this one, but it's worth discussing.
Note CHERI never gets fully disabled unlike e.g. the FPU for misa.F, so it's not an extension in the conventional sense.

This affects the priv but not unpriv spec.

arichardson, 2025-06-06:
I agree it makes sense to set this bit. The main question here is whether misa.Y indicates that CHERI instructions are available (regardless of which mode we are executing in) -- this would effectively be the mseccfg.CRE bit -- or if it it indicates the current "base ISA"/"execution mode"

I think I prefer the "are CHERI instructions available" interpretation but I don't feel strongly.
I don't like the idea of switching modes via writing to MISA because this needs to be something that is available for unprivileged code so we need the instructions anyway and having many ways of doing the same thing is not particularly appealing.

arichardson, 2025-08-06:
I agree I think we should just move mseccfg.CRE to misa.Y