riscv/riscv-cheri #636: should control flow changes to a faulty target except at the source or destination?

tariqkurd-repo, 2025-06-19:
It came out from the ARC review that faulting on branches and jumps due to a bad target is unpopular.
The proposal is to take them at the target.

One problem is how to indicate to the target what the problem is - maybe _always_ clear pcc.tag and have anothe CSR which said what went wrong?

How can we justify taking the exception at the source for security reasons?

tariqkurd-repo, 2025-06-20:
We discussed direct jumps and branches doing a representability check on the result of adding the offset to the PC, and making the result of that into the new PC, and _not_ doing the bounds check.

If the next instruction is out of bounds then it will naturally follow from the PCC check.

This conveniently removes the need to check for a "minimum sized instruction" at the target - which isn't a great check as it might be a 32-bit instruction (assuming 16-bit support) so the bounds check is effectively split in two in many cases.

For JALR, it will unseal (if sealed and the offset is zero), CADDI the offset to cs1 and redirect to the target. So again, if the result is unrepresentable, the tag will clear and so the fault will be taken at the target.

This will change some current bounds faults into tag faults, because if the offset takes the PCC out of the representable range the current spec would give a bounds fault.

Apparently Morello detectes PCC faults at the target like this

There are concerns about jumping out of a cmpartment before detecting the fault, or increasing the speculation window as a result of the proposed change.

vmurali, 2025-06-23:
> So again, if the result is unrepresentable, the tag will clear and so the fault will be taken at the target.

> There are concerns about jumping out of a compartment before detecting the fault, or potentially increasing the speculation window as a result of the proposed change.

While CHERIoT unsets the tag and takes the exception at the target (and as Wes points out, there's no security boundary crossing in existing SW platforms on a simple JALR), it might be prudent to throw the exceptions at the source for these things (untagged/sealed-but-not-zero-offset/unrepresentable/non-executable). We might very well have a CHERI system without trampolines that requires user code to clear the stack etc and use simple JALR to call and return across compartment boundaries. Fixing the ISA such that the exception is thrown only at the target will preclude such designs in the future.

nwf, 2025-06-24:
Just so it's written down somewhere...

The other aspect to this is that we apparently like to add spice to our `j{al,}r`s with CHERI.  "Sentries unseal" was probably the original sin (sorry) and is fine because it's self-contained, but then CHERIoT added IRQ-disposition-bearing sentries and we keep talking about adding indirect sentries which also commit other architectural changes as part of jumping.

It's strange, but probably not a security problem...
* if jumping to an untagged cap results in a fault "at the target" with an untagged cap installed into `mepc`.
* Similarly, if jumping to an unacceptably sealed cap raised a fault "on the target" (with `mepc` either still sealed or untagged).
* For indirect sentries, so long as IDC is clobbered with something untagged or the (still sealed) jump target register, that's probably also fine...

For `mstatus.MIE` updates, OTOH, I worry we might not be able to architecturally wait all the way until the target ifetch completes to decide whether the `mstatus` update commits.  In CHERIoT-RTOS specifically, `mstatus.MIE` is always forced on in fault handling, so it probably matters for us less than it might for other systems.

tariqkurd-repo, 2025-06-27:
https://github.com/riscv/riscv-cheri/pull/644 merged, exception checks moved to https://github.com/riscv/riscv-cheri/issues/647