riscv/riscv-cheri #638: Taggable PMAs - do we need more than one

tariqkurd-repo, 2025-06-20:
Currently in the spec we have the option of the two behaviours for storing a valid tag to a non-taggable PMA

1. Cause an access fault exception
2. Ignore the stored valid tag

We need to deal with the options.

a. choose one, and say the other is a future extension
b. add an additional PMA now

In our (Codasip's) systems we ignore the stored valid tag.
Are there any objections to changing the text to:

```
When the hart attempts to store data with the valid tag set to memory regions that are not taggable, the implementation ignores the stored valid tag.

NOTE: A PMA from a future extension could raise an access fault exception when attempting to store a valid tag to non taggable memory.
```
is anyone taking or planning to take an access fault in this case?

tariqkurd-repo, 2025-06-23:
>Trapping isn't difficult to implement though?

Correct - but I think the question is more whether you really want that behaviour in all cases in a real system.
As it's still very early in the evolution of CHERI systems I suspect that the answer is to allow both so that people aren't overly constrained.

It could be a second PMA, or a bit in (maybe) menvcfg which controls the trapping behaviour.
As PMAs may be fixed in a real system, control from a CSR bit may be desirable.

tariqkurd-repo, 2025-06-23:
>If you make it trap on stores of capabilities then you don't get to decide whether it can retire until the stored operand is known. This introduces a lot of additional complexity on most pipeline designs.

We do have this for the MMU based revocation, but that's a very different design point to a small embedded core.

> most server pipelines don't want to do precise traps for ECC store failures: they do precise traps on loads and imprecise traps on stores

RISC-V doesn't say anything about handling of ECC faults, an so I don't believe that they are mandated to be precise or imprecise. I think all store errors (AXI or ECC) can be handled asynchronously as NMIs, so I don't think this part is an issue.

I think the overriding argument is not to force embedded cores to check the tag before deciding whether they can send a store transaction to the bus.

davidchisnall, 2025-06-23:
> The cap-dirtyable machinery -- what Morello calls CDBM -- is used by the (existing) revocation machinery to close a race between 1) a core that is attempting to stop the revoker from scanning a page in subsequent epochs, and 2) a core (and its TLB) that writes a capability to that page, meaning that the revoker must continue to scan that page.

Is it sufficient in this case to clear the cap-dirty bit in the PTE and then scan the page for capabilities?  Any store of a capability concurrently with the scan will set the cap-dirty bit.

I'd be very hesitant to mandate something in the spec that is an optimisation for a research prototype of a single use case.

tariqkurd-repo, 2025-06-24:
I've made https://github.com/riscv/riscv-cheri/pull/639 have three PMA options: tagged, tag-strip and tag fault.
I'm not aware of any other sensible options, and think that it's far too soon to narrow options further than this.