riscv/riscv-cheri #669: AMO<OP>.C other than SWAP

jrtc27, 2025-07-21:
AMO{ADD,SUB,AND,(X)OR} all have their uses for capability values, and are all operations supported by C(++)11's atomics. This means atomic operations on pointer values have to be lowered to LR/SC loops when targeting CHERI, when they would be AMOs in baseline RISC-V. I believe we should provide the full set (MIN(U)/MAX(U) aren't so obviously useful but are trivial to implement once you have at least one capability AMO so should be included for completeness). ISAv9 is the same as the current proposal, but that is not because we did not think the others should exist, it is just because they're more complicated than AMOSWAP and we could get by without the others in the initial bring-up. We should however have revisited that long ago.

tariqkurd-repo, 2025-07-29:
So the suggestion is to update the address followed by SCADDR in all cases?
This is OK for cached memory but for uncached it doesn't work with CHERI atomics being supported by AXI or other bus protocols.

Is it worth proceeding with cached memory only?

tariqkurd-repo, 2025-09-05:
Given that it doesn't work for uncached memory (yet) can we postpone this one @jrtc27? What should we do?

arichardson, 2025-09-05:
IMO this should be a future enhancement and should not be required for 1.0. @jrtc27 are you okay with deferring this?

jrtc27, 2025-09-05:
Doing it post-ratification sucks because then it won't be part of the first profile and so toolchains won't be able to take advantage of the instructions. If you support capabilities in such uncached regions either your bus protocol should support these atomics or you should implement them as load-modify-store. If you can't support that then you probably can't sensibly support capabilities existing at all anyway.

arichardson, 2025-09-05:
C++20 has fetch_add/fetch_sub for std::atomic<float> and those are rarely natively supported, so I don't see this being any different.

I also don't see anyone implmenting this for uncached memory since you'd have to pass that down all the way to the target IP and that needs to understand capability arithmetic (granted it's simple for always tag-clearing, but still complex). I'd imagine having a capability AMOCAS would be more useful to avoid LR/SC loops?

I vote for having this as a fast-track extension later since there are definitely some things that need to be figured out that are not problems for the basic AMOSWAP.

tariqkurd-repo, 2026-01-29:
https://github.com/riscv/riscv-cheri/issues/923#issuecomment-3816862529