riscv/riscv-cheri #722: Should sh4add exist for non-CHERI

arichardson, 2025-08-20:
Sh4add is used sometimes in SPEC, less than the other ones, but still more frequently than e.g. XOR
```
sh1add,31761
sh1add.uw,2620
sh2add,59792
sh2add.uw,11373
sh3add,80030
sh3add.uw,16346
sh4add,11630
sh4add.uw,5111
```
Some other counts for reference
```
sll,1593
slli,32842
slli.uw,8506
slliw,4013
sllw,1067
slt,5869
slti,4907
sltiu,46735
sltu,38754
sra,42
srai,5426
sraiw,2900
sraw,470
srl,1262
srli,22619
srliw,8606
srlw,785
sub,27380
subw,15601
sw,83862
xnor,9
xor,10111
xori,16697
zext.h,7781
```

tariqkurd-repo, 2025-08-20:
IMHO that's something we can look at post ratification

arichardson, 2025-08-22:
> > That's fine, just need to make sure we didn't use the shadd4 integer encoding.
> 
> there isn't one - I repurposed the existing encoding for cap only

I meant the "natural" integer one

tariqkurd-repo, 2025-08-22:
localparam [31:0] SH1ADD             = 32'b0010000??????????010?????0110011;
  localparam [31:0] SH1ADDY            = 32'b0010000??????????011?????0110011;
  localparam [31:0] SH2ADD             = 32'b0010000??????????100?????0110011;
  localparam [31:0] SH2ADDY            = 32'b0010000??????????101?????0110011;
  localparam [31:0] SH3ADD             = 32'b0010000??????????110?????0110011;
  localparam [31:0] SH3ADDY            = 32'b0010000??????????000?????0110011;
  localparam [31:0] SH4ADDY            = 32'b0010000??????????111?????0110011;

currently 7/8 slots allocated in that group

jrtc27, 2025-08-22:
You mean 001 not 100?