riscv/riscv-cheri #785: extend mconfigptr?

tariqkurd-repo, 2025-09-15:
mconfigptr is an address but we haven't extended it

https://riscv-specs.timhutt.co.uk/spec/riscv-isa-release-f87634b-2025-07-24/riscv-privileged.html#_machine_configuration_pointer_mconfigptr_register

I guess we should..... it should be an `extended CSR`

arichardson, 2025-09-15:
I agree that this can remain as an address and software just derives the right capability on startup. Once the spec for `mconfigptr` has been finalized we could extend it?

I'd vote to remove ratification blocker. We can always extended it to be a RO capability later.

jrtc27, 2025-09-22:
Not extending it does have the sad downside that you'd either not have bounds on it or need to get bounds from somewhere else. Extending it would "just work", but the downside is that it would give you that capability to all of M-mode (or at least the part of M-mode that has ASR).

One somewhat weird option is to extend it but make its tag 0; that is, if you want to use it, you have to YBLD it with an authority you have from elsewhere.