riscv/riscv-cheri #792: YCFG's Major Encoding Is Base ISA In Disguise

jrtc27, 2025-09-18:
The current spec goes to great lengths to define a base ISA that can be simultaneously "big CHERI" and CHERIoT, and YCFG is the hack to pretend that's ok. But realistically the first thing anything targeting RVY is going to need to know is which capability encoding is in use; after all, you can't even make a >21-bit PC-relative jump (and link) without knowing the capability encoding, as that affects how AUIPC is encoded, but also anything wanting to interact with the capability bit pattern needs to know what the encoding is, which in RISC-V land would naturally be the base ISA. However, this is not how the current spec is written.

I believe that, as attractive as it is to have a single base ISA that is CHERI that everyone can build off, if the encoding in use for the base ISA with no extensions is changing between "big CHERI" and CHERIoT then we need to stop kidding ourselves and accept that means they are different base ISAs. Otherwise every RISC-V toolchain-like program is going to need to grow some -mcapability-encoding to augment -march, and that's going to suck, whereas if it's just a different -march that unambiguously encodes it in the base ISA then we slot into the ecosystem with minimal friction.

arichardson, 2025-09-18:
Now that toolchains accept profile names, I would imagine the way you'd specify it to your toolchain would be `-march=rvyb22` or `-march=cheriot-v1` and then that implies the encoding. Then the compiler can make optimization decisions based on that. IMO Having it semantically be a "base ISA in disguise" is fine to me as long as it means the toolchains can reuse almost all the shared code. If no assumptions based on the encoding are made then in theory toolchains could even generate code that is compatible with any encoding. I am not sure that is feasible though unless you don't want to pad your globals or make some very conservative assumption about representability (e.g. only powers-of-two).

To be clear, the goal here is not binary compatibility, it's source-level compatibility only.

tariqkurd-repo, 2025-09-25:
this is what we're going for for the ratificaiton package so I hope its acceptable? Please re-open if not.

jrtc27, 2025-09-25:
I don’t agree this is a sensible ISA design.