riscv/riscv-cheri #813: drootc table reference issue

Timmmm, 2025-09-29:
> [drootc](https://riscv.github.io/riscv-cheri/#drootc) is a debug mode accessible capability CSR. The address and access details are shown in [Table 23](https://riscv.github.io/riscv-cheri/#default-csrnames-added).

but if you look at Table 23 it doesn't mention anything about `drootc`?

arichardson, 2025-09-29:
Hmm let me see why this is not being rendered correctly. I think it was last touched in https://github.com/riscv/riscv-cheri/pull/731

tariqkurd-repo, 2025-09-30:
This seems to be broken  the xref is pointing to the wrong table and we don't seem to have included the correct table in the standalone spec.

So the spec doesn't actually include the CSR address anywhere.

closed 2025-09-30 by tariqkurd-repo